sfuphantom/vcu-fw #6: EEPROM Driver

gabesoares, 2020-07-30:
- [ ] Create an EEPROM driver to be able to save the VCU state periodically to flash. In the case of a power cycle or reset event, the VCU should be able to read the flash upon bootup and decide if it should load in the old vehicle state or not. 

- [ ] Ensure the driver works for both launchpad and VCU. 

- [ ] Documentation for how the driver works, and how the implementation/application works.

Junaidk11, 2020-08-10:
As a standalone FreeRTOS project, I'm able to save VCU state periodically and also at startup the task decides if VCU state to be able to initialize with default values or from the Fee Bank 7. 

The standalone project can be found here: https://github.com/Junaidk11/Launchpad_FreeRTOS/tree/freeRTOS_eepromTask

I've tested the saving and loading functionality with simulated VCU data structure update.

Junaidk11, 2020-08-15:
EEPROM task added to VCU codebase, VCU state is being saved in EEPROM bank 7 periodically - this has been tested on the launchpad. Need someone to confirm on the VCU.

Junaidk11, 2020-09-13:
Eeprom task integrated into the current VCU codebase. Able to store the VCU state periodically to flash Bank 7 (FEE) and on power-on able to read the last saved state to determine if the VCU state is to be initialized with default values or from EEPROM memory.  

Tests carried out on the launchpad, the next step is to test EEPROM on VCU and complete documentation.

Junaidk11, 2020-09-17:
EEPROM tested on the VCU and it works as expected. Moving on to documentation of the EEPROM driver and task.

Junaidk11, 2020-09-25:
Finished documentation of EEPROM driver -> explained every function, included the typical flow of EEPROM. 

Documentation can be found here: https://docs.google.com/document/d/18ksqlkycdsgcB_7pM9Fu922YyOp6K8rylBrwtv2zmuo/edit?usp=sharing

closed 2020-09-25 by Junaidk11