sfuphantom/vcu-fw #14: APPS Implausibility (Beginner)

jaypacamarra, 2021-11-24:
According to rule from 2022 Formula SAE official rules,
T.4.2.5 If an implausibility occurs between the values of the APPSs and persists for more than 100msec, the power to the (IC) electronic throttle / (EV) Motor(s) must be immediately stopped completely.
(EV only) It is not necessary to Open the Shutdown Circuit, the motor controller(s) stopping the power to the Motor(s) is sufficient.

**Objective**
Change the code in task_throttle.c such that when an APPS implausibility (APPS1 & APPS2 differs by more than 10%) occurs, we ignore all consequent throttle commands, or set THROTTLE_AVAILABLE to false. Double-check this solution with the controls team.

jaypacamarra, 2021-11-24:
Hint: You'll need to modify the logic in this function in throttle.c
[image]

rafguevara14, 2023-09-29:
This should be fixed already. We should run a quick test to validate this

rafguevara14, 2024-05-09:
stale issue

rafguevara14, 2024-05-09:
firmware handles this already. see where suspend throttle is called

closed 2024-05-09 by rafguevara14