sfuphantom/vcu-fw #16: Clean up repo

jaypacamarra, 2022-01-25:
Can we can delete these?
[image]

Can we include some of these into the .gitignore?
[image]

Why are there two .gitignores?
[image]
[image]

Guojiaxi, 2022-05-17:
I pushed a change just now to the gitignores to exclude the directories you mentioned. However, those excluded folders may still be in the repo and will need to be manually deleted from it.

Also, I also excluded the BuildConfig_* folders. Maybe this will stop the "f021.h" error.

rafguevara14, 2023-03-03:
stale issue

closed 2023-03-03 by rafguevara14