sfuphantom/vcu-fw #27: 2B: Handle the RTD procedure

kkramer215, 2023-01-07:
This function will handle the ready to drive (RTD) procedure as per the rule book requirements. When the driver flips a switch the function will check if the brake is pressed and if the tractive system is active. If both the brake is pressed and the tractive system is active the function will enable the motors to respond to inputs from the throttle. 
The brake being pressed is determined by checking if the sensor value is above a specific threshold. If the tractive system is active is determined by checking the status of the TSAL function (previously written function).

rafguevara14, 2023-03-03:
The handling of the events in the RTD procedure is complete. Now the events have to be detected via hardware interrupts

rafguevara14, 2023-09-29:
Closing and creating a new issue for the hardware part

closed 2023-09-29 by rafguevara14