sfuphantom/vcu-fw #29: 3B: Explicitely define type of FSM and unit test

kkramer215, 2023-01-07:
This is going to be a big (hopefully simplistic) redesign of the state machine. There are two parts here the state machine itself and the freeRTOS task. For the former, I want a more formal and explicit state diagram that dives into its type (Mealy or Moore) and has a proper state machine diagram. 

For the latter, the emphasis is on decentralizing fault checks. To see what I mean, [look at this](https://github.com/sfuphantom/vcu-fw/blob/state_machine_task-JJK/VCU/Phantom/tasks/source/task_statemachine.c). This task has none of its logic abstracted away by drivers. It just adds a layer of state variables that need to be constantly synched with hardware to determine its state. The events that define a certain state switch are centralized in this file. This makes it hard to see the structure of the state machine itself. This was designed from a very bottom-up approach. It is incredibly hard to tell what the inputs are and how to test them.

Instead, I think we should have a very explicitly defined set of states that get passed in as an enum to determine the state of the machine. If all fault scenarios lead to the same meaning ie APPS or BSE range fault -> SENSOR_RANGE_FAULT, then this would simplify the logic a lot. To add more fault scenarios would mean doing a check anywhere in the program and then posting a SENSOR_RANGE_FAULT event for the state machine to process.

Now from the freeRTOS side, this means that the state machine should act as a software interrupt where it immediately handles any event that is given to it. This can be done by giving it the highest priority of the tasks and ensuring its work boils down to a series of if statements (so as not to starve the processor). It can wait on an event queue to do actual work.

The outputs or consequences of the state should also be present in this task instead of spread out across the program.

rafguevara14, 2023-01-14:
This does not directly relate to the main feature of the VCU: controlling the motor by reading the foot pedal, so relative to those tasks, this should be a little lower of a priority

rafguevara14, 2023-03-03:
While the state machine is not fully complete, the general architecture described here was implemented in this PR 
https://github.com/sfuphantom/vcu-fw/commit/0e4ab76bc5661b6b59394b02642f0ff35eff44d4

closed 2023-03-03 by rafguevara14