sfuphantom/vcu-fw #31: 4A: Command Line Interface for passing throttle values

kkramer215, 2023-01-07:
Before testing real throttle petal values (APPS1, APPS2, BSE), we must ensure that the VCU code accurately processes input values. This script will be a Python command line interface for passing values from your local CMD/Terminal to the microcontroller:

This interface is tailored towards the refactor in this branch:

https://github.com/sfuphantom/vcu-fw/blob/Throttle-refactor/VCU/Phantom/Drivers/THROTTLE/README.md

kevinl03, 2023-01-31:
The script was used to send one set of values [APPS1, APPS2, BSE] at a time so that we can easily check that the VCU was receiving he values.

kevinl03, 2023-01-31:
Implementation found here: 

https://github.com/sfuphantom/vcu-fw/blob/Throttle_Simulator_Script/command-line/Throttle_Value_Simulator.py

closed 2023-01-31 by kevinl03