sfuphantom/vcu-fw #33: 4C: Software State Machine Tests

kkramer215, 2023-01-07:
This will be built on the foundation of the command line tool in VCU-4A. Its purpose is to validate the state machine task. It is only concerned with the logic (we don’t necessarily care too much about race conditions in the scope of this task). These would essentially be small “unit tests”.

This would help with the vast number of inputs to the state machine and establish some basic but integral regression tests.

Outputs:
Full design doc explaining all design decisions (we want to be very explicit about the capabilities and drawbacks of this simulation)
Be able to do something like ./run_state_tests and either fail or generate a report

rafguevara14, 2023-02-17:
Looks like there is an easier way to do this by injecting car events through the CLI.

Specifically, watch out for the `NotifyStateMachineFromISR()` in these changes

https://github.com/sfuphantom/vcu-fw/blob/Finalize-VCU-Infrastructure/Diagrams/Threads/threads.svg

rafguevara14, 2023-03-03:
PR is merged

closed 2023-03-03 by rafguevara14