sfuphantom/vcu-fw #34: 4D: VCU Simulation Mode

kkramer215, 2023-01-07:
This is the firmware side of [VCU-4A](https://docs.google.com/document/d/12nLcuveM3tNktxWxydYSn1SwOlg_8Ct3GUThzxdzUPM/edit#bookmark=id.l3vvredfgqw7). Essentially, write a single C function that does the parsing on the firmware side [here](https://github.com/sfuphantom/vcu-fw/blob/3b42a8dade1a1c27169870160567eb3f4ad83869/VCU/Phantom/tasks/source/task_throttle_agent.c#L77). 

Make sure it compiles and then we can test it in the lab in burnaby!

rafguevara14, 2023-01-14:
Need to test the final code, but it depends on @kevinl03's scripts so not a priority until they are finalized.

rafguevara14, 2023-01-19:
Created a simple encoding scheme instead of using `utf-8` to make it easier for the firmware to manage incoming messages. More details here:

https://docs.google.com/document/d/1JWH6s5qf8QeWGzkVzae5GWXK8BNsT8EGFw0WuFVnmiY/edit

This is what the pseudocode would look like from both sides:

```
# Python code 
def phantom_encode(a1: int, a2: int) -> bytes:
	value = a1 | a2 << 13
	return value.to_bytes(ceil(value.bit_length()/8), byteorder='little')

apps1, apps2 = 1062, 538
print(apps1, apps2)

message = phantom_encode(apps1, apps2)

print(f'Sending {len(message)=} packets')

# VCU Firmware code
msg = int.from_bytes(message, 'little') 
apps1_val = msg & 0b111111111111
apps2_val = (msg & 0b1111111111 << 13) >> 13

print(apps1_val, apps2_val)
```

closed 2023-02-03 by kevinl03