sfuphantom/vcu-fw #37: Fix Throttle Fault Timers

rafguevara14, 2023-02-02:
We need to figure out when these are triggered so that we can avoid them when developing and testing other features. It's also possible that these don't really work as expected...

https://github.com/sfuphantom/vcu-fw/blob/a6cba0043ec8ddae5682d43976287c164753363d/VCU/Phantom/tasks/source/task_throttle_actor.c#L110

kkramer215, 2023-02-14:
Best to figure this out ASAP to avoid trying to "fix" other code that might be impacted by the timers. Would be good to try to isolate the timers to see if working properly or not.

rafguevara14, 2023-02-17:
Looks like this gets fixed by the new architectural design. will link PR once it's up!

https://github.com/sfuphantom/vcu-fw/blob/Finalize-VCU-Infrastructure/Diagrams/Threads/threads.svg

rafguevara14, 2023-03-03:
PR is merged

closed 2023-03-03 by rafguevara14