sfuphantom/vcu-fw #41: Handle CAN messages

rafguevara14, 2023-05-09:
When CAN messages are received either through interrupts or polling (to be decided by the CAN/Networking team), the firmware will need to dispatch and redirect the messages to the appropriate thread to handle and/or store the information.

rafguevara14, 2023-05-23:
depends on the CAN team defining all can messages first

kevinl03, 2026-01-17:
There are a few steps required to handle **CAN messages**, split into two parts:

* **Upstream** = sending side
* **Downstream** = receiving side

### Reference (TMS570 CAN Driver Setup)

In **Aug 2024**, @danrobren and @kevinl03 validated the CAN drivers using this tutorial to set up CAN communication on the **TMS570**:
[https://www.youtube.com/watch?v=TrKQN-sor-s](https://www.youtube.com/watch?v=TrKQN-sor-s)

### Hardware Requirements

The next person taking this task should:

* get access to the **microcontroller hardware from the lab**

### Recommended Software Setup

On the software side, it’s recommended to have the following installed/configured:

* **Code Composer Studio (CCS)**
* **HalCoGen** (for pin + peripheral configuration)

### System Context

From my latest knowledge, CAN communication is used across multiple modules in the EV. The following diagram shows the main sections where CAN is used:

[image]

---