sfuphantom/vcu-fw #44: Output relevant signals during a severe fault scenario

rafguevara14, 2023-05-09:
I'm being vague because there may be other things to consider, but I think we have to notify the SDC in this event

rafguevara14, 2024-07-12:
For testing, as discussed with @juliannesoriano20 , check that when any of the shutdown events are triggered, the logs or a voltage is seen on the pin

juliannesoriano20, 2024-07-12:
Should VCU_FLT be kept high until the car is reset? I think it is good to keep VCU_FLT high whenever we have a fault, that way SDB is sure to act upon the signal. Then, only turn low once the car is reset and then we can set VCU_FLT low

rafguevara14, 2024-07-12:
> Should VCU_FLT be kept high until the car is reset? I think it is good to keep VCU_FLT high whenever we have a fault, that way SDB is sure to act upon the signal. Then, only turn low once the car is reset and then we can set VCU_FLT low

I guess it depends on how the shutdown board is designed. Doublecheck if the shutdown board latches the signal based on this schematic

rafguevara14, 2024-07-12:
So for now, you can assume we have the reset signal available for us and do what you suggested (turn vcu fault low on a reset). Make a comment in the code about this assumption (eg TODO: Clarify if we have reset signal)

kevinl03, 2026-01-17:
Looks like this issue was complete with the coolant logic

### Reference PR (How to Add a Test Event / Command)

Use this PR as the template for implementing an event-based testing hook (coolant logic PR):

* [https://github.com/sfuphantom/vcu-fw/pull/73/files](https://github.com/sfuphantom/vcu-fw/pull/73/files)

---

Can still reference the above comments to reproduce desired behaviour