sfuphantom/vcu-fw #45: Implement BOTS logic

rafguevara14, 2023-05-19:


rafguevara14, 2023-05-19:
Blocked waiting on updates for Power.

rafguevara14, 2024-05-09:
@rubberduck653 Assign to justin once he is in the github

jykh24, 2024-06-01:
T.3.3 Brake Over Travel Switch - BOTS 
T.3.3.1 The vehicle must have a Brake Over Travel Switch (BOTS). Brake pedal travel exceeding the  normal range will actuate the switch 
T.3.3.2 The BOTS must be a mechanical single pole, single throw (commonly known as a two position)  switch (push-pull or flip type).
Formula SAE® Rules 2024 © 2023 SAE International Page 63 of 140 Version 1.0 1 Sep 2023 
T.3.3.3 Operation of the BOTS to the OFF position must Open the Shutdown Circuit IC.9.2.2 / EV.7.2.2 T.3.3.4 Repeated operation of the switch must not reset or restore power  
T.3.3.5 The driver must not be able to reset the BOTS.  
T.3.3.6 The BOTS must be implemented with analog components, and not using programmable logic  controllers, engine control units, or similar functioning digital controllers.

jykh24, 2024-06-02:
The BOTS signal will be low until activated and then should remain high and send a signal to open the shutdown circuit

jykh24, 2024-06-08:
update task_throttle.c to include logic for clearing bots logic after receiving clearing signal from single throw switch